ArduPilot/MissionPlanner #2579: NuGet slowing down build on VS 16.9.2

custeve, 2021-03-26:
I am able to build, but it's very slow. It would appear that NuGet is attempting to fetch packages on every build. I'm currently attempting to modify NuGet based on some advice here: 

https://stackoverflow.com/questions/31180942/nuget-updates-packages-during-build-even-when-configured-to-not

I believe that once packages are restored, the actual compile is quite quick. If there are any other tips to speed up compiling, that'd be great. 

Thanks!

custeve, 2021-03-26:
FYI - I was able to resolve this by disabling NuGet from downloading missing packages after the first successful build. Not sure this is the best answer, but it works for now.

meee1, 2021-04-12:
this is a VS issue, so i asume it will get fixed, and closing for now

closed 2021-04-12 by meee1