ArduPilot/MissionPlanner #2582: Issue Opening Main Form Designer in Version 1.3.74

RKRCyphy, 2021-03-30:
I have cloned the latest version of the code and was able to compile and run the application. The issue i have is when i try to open the designer of  form MainV2.cs , it shows MissingMethodException. Please find attached screenshot of the same.

Please advice on further steps.
[image]

RKRCyphy, 2021-03-31:
From the previous error i tried opening designer for myButton.cs in project ExtLibs->WinForms->MissionPlanner.Controls.
Attached is the error screenshot for the same. 

Please advise i am not able to move ahead on this.
[image]

meee1, 2021-04-12:
at a guess update your  VS version