ArduPilot/MissionPlanner #2598: UBlox GPS AssistNow support

rmackay9, 2021-04-30:
We can improve GPS lock time and position estimate quality (at least soon after startup?) by adding support for Ublox's Assist Now feature.  This involves:

- [ ] applying for a key from UBlox
- [ ] write a script to be run on firmware.ardupilot.org to download the almanac data from UBlox every hour or so
- [ ] enhance MP and other ground stations to download the latest almanac and forward the data onto the autopilot
- [ ] enhance AP to forward the almanac data received from the GCS to the GPS

Corresponding ArduPilot flight code issue: https://github.com/ArduPilot/ardupilot/issues/17305

meee1, 2021-04-30:
i used to have something like this
https://github.com/ArduPilot/MissionPlanner/blob/11688c6230b098b6682316174d937931ce28ee3d/MainV2.cs#L2602

but was removed some time ago, as the time to first fix on most gps's is under 60 seconds these days

Sebbb, 2021-05-01:
> but was removed some time ago, as the time to first fix on most gps's is under 60 seconds these days

For me, sometimes it takes up to 10 minutes to get a reliable fix with >12 satellites and hdop <0.8. Especially as I live in a city center and can only drive outside to fly on weekends. With my old GPS I was able to speed it up with ublox AssistNow. With my new one, as it's a Matek M9N-CAN (onboard GPS, running AP_Periph, connected via CAN) without exposed UART. that does not work anymore. Hence the request/question at discuss.ardupilot.org :)