ArduPilot/MissionPlanner #2605: When I go simulation -> copter on windows - The sitl spits out a json error.

MichaelSwan3, 2021-05-04:
#### Issue details
_Please describe the problem, or desired feature_
When I go simulation -> copter on windows
The sitl spits out this error
[image]
Unexpected character encountered while parsing value.
Have
* reinstalled mission planner.
* Ran the same thing on a friend windows computer and it gave the same error.

It looks like there has been a change in the soft in the loop, where is parses some json and someone has changed the json to throw an error.

#### Version
_What version was the issue encountered with_
1.3.74

#### Platform
[  ] All
[  ] AntennaTracker
[ x ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
hex

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
software only

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2021-05-04:
please try beta MP

help > update beta

framosnt, 2021-05-11:
any idea to continue using mission planner 1.3.72 and simulator ???

JulioCesarMatias, 2021-05-15:
> any idea to continue using mission planner 1.3.72 and simulator ???

you can solve the problem by upgrading to the beta version
[image]

casrya, 2021-05-16:
The problem is that the downloaded **vehicleinfo.py** has unquoted uppercase boolean literals True which are not valid JSON and so should really be fixed at the ArduPilot side of things IMO.

Anyway, there is no easy fix other than modifying the code since vehicleinfo.py is always refreshed via download.

In case it is helpful to anyone I have attached a screenshot of my own fix since I am using a heavily modified 1.3.74 which I don't want to merge with the latest until I have a really good reason to ;)

In SITL.cs
[image]