ArduPilot/MissionPlanner #2609: MP can install silently

G0rocks, 2021-05-07:
#### Issue details
_Please describe the problem, or desired feature_
Feature request.
The idea is to be able to install mission planner silently so installation is simpler and so you can add the installer to package managers like [winget](https://github.com/microsoft/winget-cli).
I was trying to add MP to the winget packages but I ran into this problem when MP didn't install silently.
Any help appreciated, thanks :)

#### Platform
[x] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

lebarsfa, 2021-06-27:
FYI I just [submitted](https://community.chocolatey.org/packages/missionplanner/) a [Chocolatey](https://chocolatey.org/install) package, it can be installed silently (without the drivers, which we could later try to provide as a separate package, but it seems harder to make it silent) with 
```powershell
choco install -y -r --no-progress missionplanner --version=1.3.74
```
After validation by the moderators, the version parameter can be removed to just install the latest.

G0rocks, 2026-09-10:
I have since this issue uploaded a program that does not install silently to winget so I will try again

G0rocks, 2026-09-10:
Here is the new attempt:
https://github.com/microsoft/winget-pkgs/pull/432539

G0rocks, 2026-10-29:
I will rename this issue from "MP can install silently" to "add MP to winget-pkgs".

G0rocks, 2026-10-29:
The PR for adding to winget was closed because the installer stopped and did not complete during the installation timeout of 3 hours during testing.
See https://github.com/microsoft/winget-pkgs/pull/432539