ArduPilot/MissionPlanner #2611: Getting Direct To Waypoint angle via IronPython

berkaydedeoglu, 2021-05-08:
### Issue details
We can use Python via IronPython then we can reach lots of data which we can already oberve on screen. However; I am not sure that we can retrieve details about the line which named as 'Direct to WP' and colored with orange. Is there any way to get some details about that line? Especially i wonder if i can get the angle of this direct according to a reference environment.

That one is not really an issue maybe this may be evaluated as a feature idea. 

### Version
v1.3.11

### Platform
[+] Rover

### Hardware type
PixHawk

meee1, 2021-05-08:
MAV.cs.yaw,
                    MAV.cs.groundcourse, MAV.cs.nav_bearing, MAV.cs.target_bearing,
       
It's one of these

berkaydedeoglu, 2021-05-10:
"MAV.cs.target_bearing" is what i totally expected.
Yes there are already some commands as you said. I didn't realized them bcoz they looks like static variables.

closed 2021-05-10 by berkaydedeoglu