ArduPilot/MissionPlanner #2612: Remove "Change Alt" button for Copter and Rover

rmackay9, 2021-05-09:
[Users sometimes ask about the "change alt" button](https://discuss.ardupilot.org/t/change-altitude-under-action-tab/70848) but it isn't support in Copter, Rover or Sub so we should remove it for these vehicles to reduce user confusion and support calls.

[image]

rmackay9, 2025-05-15:
Hi @mrcdnov,

It probably uses https://mavlink.io/en/messages/common.html#MAV_CMD_CONDITION_CHANGE_ALT

jagrandle, 2025-05-17:
Alternatively, could we make it so that this button works as intended ? It can be very useful in Copter to be able to simply command a climb or decent to a new altitude ?

It seems from the message above that there might already be function available that it could link to.

Is it even possible for Mission Planner to display different options in this Actions tab based on the vehicle type ? If so, could we possibly change the default value for copter Change Speed to something like 10 m/s instead of the 100m/s currently listed ? It would make a lot more sense, and is unlikely to have any catastrophic consequences for a multirotor, compared to commanding 100m/s accidentally !

Thanks