ArduPilot/MissionPlanner #2614: UAVCAN SLCAN update firmware not working

CarlEngel, 2021-05-12:
#### Issue details
I am trying to do an update a pixhawk 2.1 running ardupilot in SLCan mode. I am using local firmware. When I click update the progress bar appears and quickly disappears. The log is reporting the device name does not match (can be seen below). However, even the log message shows the exact same device names which leads me to think that there is something wrong with the white space. I have tried changing the device name a few times with no difference.

When inspecting the UAVCAN bus I notices that it does not even send the uavcan.protocol.file.BeginFirmwareUpdate message. Which means it is getting stopped by mission planner before hand. I tracked the problem down to ExtLibs\UAVCAN\UAVCan.cs however I am not familiar enough with mission planner or C# to try fix it.

#### Version
Mission Planner 1.3.74
ArduCopter V4.0.7 (0bb18a15)

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Hardware type
Pixhawk 2.1 CubeBlack 00330034 31375105 

#### Logs
This is from the mission planner console log:

Update 125 E881Coax  0 C:\Users\Documents\E881Coaxuavcan.bin
125 GetNodeInfo: seen 'E881Coax' from 125
125 device name does not match E881Coax  vs E881Coax
Update EXIT

meee1, 2021-05-12:
Can you give me a log.   Before connecting there is a log button in the top right of the screen. Tick that and then click the. Can1 connect button. And try the update.  After that please post the .can file in the mission planner log directory

meee1, 2021-05-13:
its failing because of this

device name does not match E881Coax vs E881Coax

i asume there might be some non printable characters thats causing the check to fail. this is the reason for the log, to verify this

CarlEngel, 2021-05-13:
Here are all the logs from when it is happening.

[2021-05-13 10-30-18.zip](https://github.com/ArduPilot/MissionPlanner/files/6473089/2021-05-13.10-30-18.zip)

meee1, 2021-05-19:
that log doesnt have apear to have a firmware update request

meee1, 2021-05-19:
[2021-05-13 10-30-18.txt](https://github.com/ArduPilot/MissionPlanner/files/6506288/2021-05-13.10-30-18.txt)

CarlEngel, 2021-05-19:
That is correct, it is not even trying to send the request, it gives this error before it sends the request.