ArduPilot/MissionPlanner #2620: preload parameters from a file

tridge, 2021-05-24:
It would be very nice to have a way to preload parameters from a file, so param download is skipped.
The reason is for this sort of flight:

- takeoff is using one GCS
- handover to another GCS at long range and high altitude
- new GCS param download will happen on a very poor link
- although param download can be cancelled, this precludes param changes in flight (eg. set AFS_TERMINATE)

davidbuzz, 2021-05-25:
thoughts..https://github.com/ArduPilot/MissionPlanner/pull/2623  ?   its enough of an implementation to prove it works.  needs to cleanup the .pck file  at some point, but doesn't right now.

meee1, 2021-05-26:
is caching good enough. or the ability to just load a known set the target?

davidbuzz, 2021-05-26:
> is caching good enough. or the ability to just load a known set the target?

Not sure i understand the difference..?  Once u have the .pck file cached to disk one one computer, copying it to the others at the same location,  so they can all read from the same file is what I was thinking with the pr.

meee1, 2021-05-27:
one affects everyone.
the other is for a load these now, and ignore any other action