ArduPilot/MissionPlanner #2631: Plan: Fence Radius truncated to integer

rmackay9, 2021-06-01:
The Plan screen's Fence list has a "Radius" parameter which accepts floating point numbers but seems to truncate them to integers.  The issue can be reproduced by doing this:

- Open MP and start the Simulator (Rover or Copter)
- Open the Plan screen and select FENCE from the drop-down on the top right
- Click on the map in a few places to create FENCE_CIRCLE_EXCLUSION fences (BTW, I think these string could be made more readable by changing to "Circle Exclusion" but I think I've already raised another issue about this)
- Change a radius to "3.5" and press "Write"
- Push the "Read" button to read back the fences and note that "3.5" has been changed "3".

[image]

I suspect that MP is changing this field to an integer because it is normally used for the "p1" parameter of commands which is always and integer.  From what I can see the MAV_CMD_NAV_FENCE_CIRCLE_EXCLUSION message accepts floating points but please tell me if I'm wrong.

meee1, 2021-06-01:
i believe this is AP dropping the decimals

meee1, 2021-06-01:
[image]

peterbarker, 2021-06-01:
Found the bug.  It's in ArduPilot.  Closing this.

closed 2021-06-01 by peterbarker