ArduPilot/MissionPlanner #2635: Plan, Fence Save/Load created "UNKNOWN" fence point

rmackay9, 2021-06-05:
There seems to be an issue either saving or loading fence point to/from a file.

To recreate the issue create a fence as shown below:
[image]

Then "Save File" and then immediately "Load File" and the result is the original fence list with an additional "UNKNOWN" at the top
[image]

Ryanf55, 2025-03-26:
I just tested on `master`, and it's worse, the points don't load, it throws immediately.

[image]

EosBandi, 2025-03-26:
The original bug was fixed ages ago. 
The error you described only happens when you trying to load fence while the planner is not in Fence mode. 

[image]

rmackay9, 2025-03-26:
OK, if this is already fixed then let's close this issue.  Thanks @EosBandi !

closed 2025-03-26 by rmackay9