ArduPilot/MissionPlanner #2667: fatal: remote error: upload-pack: not our ref

gitfishup, 2021-08-05:
First，  git pull upstream master,
Then   git Submodule update --init --recursive --force  ， error occurs： 

fatal: remote error: upload-pack: not our ref 478c5bb4b40abd1ceabbe3d1f14ea40ef505f7c4
fatal: remote error: upload-pack: not our ref a17eb357235e4f6d409cdd664daceda2fcac1e7b

git log 

[image]

git Submodule update --init --recursive --force  ， error occurs： 
[image]

Can anyone help me ?

meee1, 2021-08-05:
Don't do recursive in the submodule

gitfishup, 2021-08-05:
> Don't do recursive in the submodule

Oh，Thanks a lot.

closed 2021-08-05 by gitfishup