ArduPilot/MissionPlanner #2668: NETSDK1045	The current .NET SDK does not support targeting .NET 6.0.

gitfishup, 2021-08-05:
#### Issue details
Error	NETSDK1045	The current .NET SDK does not support targeting .NET 6.0.  Either target .NET 5.0 or lower, or use a version of the .NET SDK that supports .NET 6.0.	wasm	C:\Program Files\dotnet\sdk\5.0.302\Sdks\Microsoft.NET.Sdk\targets\Microsoft.NET.TargetFrameworkInference.targets	

[image]

vs2019 version 
[image]

What 's going on?

gitfishup, 2021-08-06:
go to https://dotnet.microsoft.com/download/dotnet/6.0 to download a dotnet-sdk-6 for vs2019 16.10 , install it ,then the problem is solved.

[image]

closed 2021-08-06 by gitfishup

MLEI-AERO, 2021-08-24:
I still have the same problem. Installing .NET 6.0.0-preview for my VS2019 16.11.0 doesn't solve it. Do I have to set a path or anything in VS?

[image]

gitfishup, 2021-08-25:
> I still have the same problem. Installing .NET 6.0.0-preview for my VS2019 16.11.0 doesn't solve it. Do I have to set a path or anything in VS?
> 
> [image]
@MLEI-AERO I solved the problem at the beginning.But the problem came up again the next day.If vscode prompts you to update something, restart after the update.If you haven't updated anything, try building a new missionplanner.After installing NET, if the problem still occurs, this is how I solve it

danilopmaciel, 2022-07-27:
Same problem 
[image]

meee1, 2022-07-28:
install dotnet 6