ArduPilot/MissionPlanner #2670: Horizonal Accuracy gpsh_acc -1 with wireless telemetry

jondave, 2021-08-09:
#### Issue details
I have a Pixhawk 4 when connected with a USB cable directly to my computer mission planner gives a value of horizontal accuracy / gpsh_acc (usually around 0.9 to 1.5 outside) but when connected via wireless telemetry the horizontal accuracy is always -1.
In Mavlink inspector gpsh_acc is always 0 when connected with wireless telemetry but is -1 in the Quick view panel.
The correct gpsh_acc values are recorded in the data logs when connected either via USB or wirelessly.

The same problem exists for the Holybro Pixhawk 4 GPS modules and for the Holybro H-RTK F9P Rover Lite GPS modules.

#### Version
Mission PLanner 1.3.75 build 1.3.7883.26333
ArduRover V4.0.0

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[ X] Rover
[  ] Sub

#### Airframe type
Ground Rover

#### Hardware type
Pixhawk 4
Holybro Pixhawk 4 GPS modules
Holybro H-RTK F9P Rover Lite GPS modules

#### Logs
Logs show the correct values.
[image]

[image]

meee1, 2021-08-09:
update the FW on your telem radios. and also check you have mavlink 2 enabled on that port

jondave, 2021-08-09:
Thanks that has worked by changing to mavlink 2.