ArduPilot/MissionPlanner #2678: NU1104:Please check if the project reference is valid and if the project file exists.

JOJObail, 2021-08-20:
#### Issue details
NU1104 Cannot find the project "E:\Drone\MissionPlanner-master\MissionPlanner-master\ExtLibs\mono\mcs\class\System.Windows.Forms\System.Windows.Forms-net_4_x.csproj".Please check if the project reference is valid and if the project file exists.BaseClasses E:\Drone\MissionPlanner-master\MissionPlanner-master\ExtLibs\BaseClasses\BaseClasses.csproj 1

#### Version
VS2017  15.98

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

meee1, 2021-08-20:
Checkout the submodule

JOJObail, 2021-08-20:
> 检查子模块

Thanks. But I see from the source code on the website that there is no mono folder in ExtLibs, and there will be an extra mono folder in the compressed package I downloaded, and there is nothing in it.

JOJObail, 2021-08-23:
> Checkout the submodule

When I update the submodule and enter "git submodule update --init --recursive" in git, git does not respond, skips directly, and does not report an error

meee1, 2021-08-23:
sounds like a problem with your git install.

JOJObail, 2021-08-23:
> sounds like a problem with your git install.

But clone repo succeeded

meee1, 2021-08-23:
[image]
works just fine for me

JOJObail, 2021-08-23:
My display is like this.
[image]

meee1, 2021-08-23:
something wrong with your git

JOJObail, 2021-08-23:
> something wrong with your git
When I updated, I ran into such a problem
[image]

danilopmaciel, 2022-07-26:
[image]
The same problem. I get zip file unzip and compile but the problem persist.

danilopmaciel, 2024-11-19:
[image]
problem again using gitclone from vs community