ArduPilot/MissionPlanner #2681: I still have the same problem. Installing .NET 6.0.0-preview for my VS2019 16.11.0 doesn't solve it. Do I have to set a path or anything in VS?

MLEI-AERO, 2021-08-24:
I still have the same problem. Installing .NET 6.0.0-preview for my VS2019 16.11.0 doesn't solve it. Do I have to set a path or anything in VS?

[image]

_Originally posted by @MLEI-AERO in https://github.com/ArduPilot/MissionPlanner/issues/2668#issuecomment-904601306_

meee1, 2021-08-24:
Compile just missing planner project. You don't need wasm for mission planner to run

meee1, 2021-08-24:
The answer to the actual issue here though is install VS 2022 preview

MLEI-AERO, 2021-08-24:
Sorry for german VS code. However, when you say just compile mission planner you mean I should ignore errors and just press run (green arrow). As I understand only MissionPlanner is selected?

[image]

closed 2021-08-25 by MLEI-AERO