ArduPilot/MissionPlanner #2688: add magic Alt-A as button on Frame Type page

tridge, 2021-09-15:
discussed on dev call, thought adding the nice Alt-A config screen as a button on this page would be nice:

[image]

button could be called "Launch Frame Config Tool", or maybe popup asking if they want to launch it after setting FRAME_CLASS?

it would be great for quadplanes too ... needs to be Q_ params

EosBandi, 2021-11-09:
I just added a PR that revamps and moves the InitialParameter Calculator to the Setup tab in MP.
https://github.com/ArduPilot/MissionPlanner/pull/2723

EosBandi, 2021-11-15:
PR was merged, So Instead of Alt-A, the initial parameters calculator now resides in the Setup tab, right after the frame configuration. If there are no recommendations to the params calculator, then this PR can be closed.
[image]