ArduPilot/MissionPlanner #2709: button to restart lua scripts

tridge, 2021-10-29:
when doing lua devel it is a pain to restart SITL all the time. We need this command:
https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/modules/mavproxy_misc.py#L342
just need a "Lua Restart" button in control-F I think

patrickpoirier51, 2022-02-07:
@meee1   yes please, this feature would be so useful when tuning script on real scenario as well.
My suggestion is to add a RESTART option when you select file on the SCRIPT directory so you can select a specific script to restart.

meee1, 2022-02-07:
already exists in the actions tab.  i might need to widen the display though

patrickpoirier51, 2022-02-07:
@meee1  ok just found it  :-)
[image]

meee1, 2022-02-07:
you will need to do an update or beta im releasing now, i double checked, and it was broken

patrickpoirier51, 2022-02-07:
Yes, I will try later, Thanks

closed 2022-07-20 by tridge