ArduPilot/MissionPlanner #2711: Log Browser: map adds lines between waypoint and the last waypoint (sometimes)

rmackay9, 2021-11-01:
While analysing logs from my recent mapping boat test from Shimane, I've noticed that the Log Browser screen draws lines not only between waypoints (which is very helpful) but also between the waypoint and the last waypoint but only sometimes.  It seems related to whether the waypoint was moved.

Below is an example where I moved waypoints 7, 8, 11 and 12 away from the shoreline for additional safety.  This resulted in extra lines appears between home and 11 and 12.  I'm unsure why 7 and 8 didn't also get lines to home.
[image]

I've placed the log file here but it is quite big sorry!: https://www.dropbox.com/s/ijgazuvakcfstlc/00000050.zip?dl=0

EosBandi, 2022-01-03:
It is indeed because of changing mission en route. The easiest fix is to do not draw the lines for changed waypoints, just simply put the updated markers to the map. It will look like this 
[image]
Is it OK ?

rmackay9, 2022-01-03:
Thanks!  That looks better to me but I'll leave it to @meee1 to decide what is the best things to do.