ArduPilot/MissionPlanner #2713: Send `MAV_CMD_PREFLIGHT_CALIBRATION` param 4 for RC calibration.

IamPete1, 2021-11-01:
https://mavlink.io/en/messages/common.html#MAV_CMD_PREFLIGHT_CALIBRATION

Thanks to https://github.com/ArduPilot/ardupilot/pull/18514 AP will now refuse to arm in RC calibration. Mission Planner should send MAV_CMD_PREFLIGHT_CALIBRATION param 4 = 1 when RC calibration is started and param 4 = 0 when calibration is completed.

meee1, 2021-11-01:
@IamPete1 whats the intent behind this change?

MP currently reads the pwm values and uses these to set the min/max/trim

is this 100% moving it to the AP?

IamPete1, 2021-11-01:
@meee1 No other change required on the MP side. This new bit just stops the vehicle accidentally arming while the user is moving there sticks to the extremes.