ArduPilot/MissionPlanner #2717: failed to load fw when running px4 bootloader

tridge, 2021-11-05:
when a user buys a board that came with px4 and wants to use ArduPilot they will want to use MissionPlanner to load the firmware. That doesn't work at the moment, I suspect because of the recent extflash support. I applied a fix for this to Tools/scripts/uploader.py that I think needs to be applied in MissionPlanner as well.
This was the fix needed for uploader.py:
https://github.com/ArduPilot/ardupilot/commit/dfd26868780084148bdb8e833fd5f33a05175e8c#diff-c16aa82ddeb6577c91570778032de965e3c728dfd7dc668abb2bf7a03368702f
There are two key changes:
 - check for ext flash size before the board_type, board_rev and fw_maxsize checks
 - resync after a failed extflash check

Here is what happens with a new out of the box Pixhawk5x.
[image]

tridge, 2021-11-05:
more debug, I added some print() calls to show leading byte (command code) in each send[]
[image]
the sequence of commands looks fine .... don't know why it doesn't accept it

meee1, 2021-11-05:
Do we know what bootloader code version is on it?

tridge, 2021-11-06:
> Do we know what bootloader code version is on it?

it says blrev 5, but that doesn't say much

tridge, 2021-11-06:
i've tried lots of things to debug with no luck. You can see my various hacks here:
https://github.com/tridge/MissionPlanner/commits/pr-upload-px4-bl
(I use CI to do the MissionPlanner build as I've never managed to get builds working on my win10 box)
another data point: fw load of ardupilot works fine with qgc

closed 2021-11-07 by tridge