ArduPilot/MissionPlanner #2737: Reboot button

mrnams, 2021-12-02:
#### Issue details
Like iNav or Betaflight it should have reboot button or (save and reboot), and make at common place where it can be accessed from all tabs may be before ARDUPILOT logo.

#### Version
_What version was the issue encountered with_
All versions
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Not applicable

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Not applicable

#### Logs
_Please provide a link to any relevant logs that show the issue_
Not applicable

EosBandi, 2021-12-03:
Press CTRL-F and press the Reboot Pixhawk button.
I think any more prominent place is potentially dangerous.

meee1, 2021-12-03:
there is  "preflight reboot" in the actions tab, this is checked before just doing though.

but in general i agree with @EosBandi

mrnams, 2021-12-04:
@meee1 @EosBandi  Yes we can do reboot using CTRL+F, but that's not user friendly, just comparing with iNav,betaflight they have user friendly button "Save and Reboot"

meee1, 2021-12-04:
i would suggest using what i suggested
on Flight Data > Actions Tab > you can change the Do Action to "Preflight Reboot" to do this.

mrnams, 2021-12-07:
Yes we can use it,my intention was just to make it simpler, but at same time i understand complexity of mision planner compared with betaflight and iNav configurator.
May be we can think of this whenever we plan to redesign it UI in future.