ArduPilot/MissionPlanner #2742: 'New Firmware' notification causes Mission Planner to crash when using Mono

j-chen-opteran, 2021-12-09:
#### Issue details
When running Mission Planner on Linux using `mono MissionPlanner.exe`, if the vehicle firmware is not the latest version, Mission Planner crashes as soon as a UDP connection is established. 

I am running Copter 4.1.1. After establishing a connection over UDP, I get a popup saying that Copter 4.1.2 is available. As soon as this happens, Mission Planner crashes. 

#### Version
1.3.75 build 1.3.7883.26333

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Quad H

#### Hardware type
Matek H743

#### Logs
```
[xcb] Too much data requested from _XRead
[xcb] This is most likely caused by a broken X extension library
[xcb] Aborting, sorry about that.
mono: ../../src/xcb_io.c:741: _XRead: Assertion `!xcb_xlib_too_much_data_requested' failed.

=================================================================
	Native Crash Reporting
=================================================================
Got a SIGABRT while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

=================================================================
	Native stacktrace:
=================================================================
	0x55fc99584c4b - mono : 
	0x55fc99584fdd - mono : 
	0x55fc99531e47 - mono : 
	0x55fc995841cc - mono : 
	0x7f138154b3c0 - /lib/x86_64-linux-gnu/libpthread.so.0 : 
	0x7f138103c18b - /lib/x86_64-linux-gnu/libc.so.6 : gsignal
	0x7f138101b859 - /lib/x86_64-linux-gnu/libc.so.6 : abort
	0x7f138101b729 - /lib/x86_64-linux-gnu/libc.so.6 : 
	0x7f138102cf36 - /lib/x86_64-linux-gnu/libc.so.6 : 
	0x7f1377a0a415 - /lib/x86_64-linux-gnu/libX11.so.6 : _XRead
	0x7f1377a0af95 - /lib/x86_64-linux-gnu/libX11.so.6 : _XRead32
	0x7f13779ec086 - /lib/x86_64-linux-gnu/libX11.so.6 : XListFontsWithInfo
	0x7f1377a43599 - /lib/x86_64-linux-gnu/libX11.so.6 : 
	0x7f13779f8873 - /lib/x86_64-linux-gnu/libX11.so.6 : XCreateOC
	0x7f13779ecb82 - /lib/x86_64-linux-gnu/libX11.so.6 : XCreateFontSet
	0x40cd841f - Unknown

=================================================================
	Telemetry Dumper:
=================================================================
Pkilling 0x139720199763712x from 0x139721516766976x
Pkilling 0x139720217282304x from 0x139721516766976x
Pkilling 0x139720988288768x from 0x139721516766976x
Pkilling 0x139719375582976x from 0x139721516766976x
Pkilling 0x139721726899968x from 0x139721516766976x
Pkilling 0x139720219383552x from 0x139721516766976x
Pkilling 0x139720990390016x from 0x139721516766976x
Pkilling 0x139721524954880x from 0x139721516766976x
Pkilling 0x139721745321920x from 0x139721516766976x
Pkilling 0x139720970077952x from 0x139721516766976x
Pkilling 0x139719610652416x from 0x139721516766976x
Pkilling 0x139720972179200x from 0x139721516766976x
Pkilling 0x139720974280448x from 0x139721516766976x
Pkilling 0x139721512564480x from 0x139721516766976x
Pkilling 0x139720976381696x from 0x139721516766976x
Pkilling 0x139719595431680x from 0x139721516766976x
Pkilling 0x139721514665728x from 0x139721516766976x
Pkilling 0x139721477945088x from 0x139721516766976x
Pkilling 0x139720966272768x from 0x139721516766976x
Pkilling 0x139720978482944x from 0x139721516766976x
Pkilling 0x139719597532928x from 0x139721516766976x
Pkilling 0x139720980584192x from 0x139721516766976x
Pkilling 0x139721518868224x from 0x139721516766976x
Pkilling 0x139721510381312x from 0x139721516766976x
Pkilling 0x139720982685440x from 0x139721516766976x
Pkilling 0x139719369279232x from 0x139721516766976x
Pkilling 0x139721520969472x from 0x139721516766976x
Pkilling 0x139720213079808x from 0x139721516766976x
Pkilling 0x139719371380480x from 0x139721516766976x
Pkilling 0x139720197662464x from 0x139721516766976x
Pkilling 0x139720203958016x from 0x139721516766976x
Pkilling 0x139720215181056x from 0x139721516766976x
Pkilling 0x139719373481728x from 0x139721516766976x
Entering thread summarizer pause from 0x139721516766976x
Finished thread summarizer pause from 0x139721516766976x.
Failed to create breadcrumb file (null)/crash_hash_0xd73c41f0e

Waiting for dumping threads to resume
bps 9316 loss 0 left 0 mem 97.6748046875 mav2 True sign False mav1 0 mav2 171 signed 0      
```

mych907, 2022-01-07:
I have the same issue. Mission planner crashes with notification "New Firmware Update Copter 4.1.3"

RoboViz, 2022-06-08:
Same issue, 4.1.5 on copter 4.2.1 current stable version

Xploror, 2022-06-13:
Did anyone solve this problem, even Im facing it too

Jazhi, 2022-07-13:
I faced the same issue. By the way, the [README.md](https://github.com/ArduPilot/MissionPlanner) mentioned that Linux requirements include "libmono-winforms4.0-cil" which is not available for Ubuntu 20.04. Not sure if it's related or not.

TheLoki2020, 2022-09-07:
I'm getting the same issue. Any news ?

rtalans, 2022-10-18:
I'm burnt out looking for an answer to this. Is anyone checking this thread?

Transistorsis, 2022-10-19:
I haven't posted my workaround yet as I was hoping this would get fixed instead. But, as that doesn't seem to be happening, here's my workaround:

Because the crash is due to something around checking if there is new firmware available, all you need is to disable mission planners ability to perform this check. I've done this by disabling internet access to mission planner. This does have some downsides though. One is that you can't upgrade the firmware without internet access for some reason.

Using some information from [here](https://serverfault.com/questions/550276/how-to-block-internet-access-to-certain-programs-on-linux). 

1. Create a new linux user group `groupadd no-internet`
2. Add your user to the group `useradd -g no-internet <your-username>`
3. Run mission planner with the command `sg no-internet "mono /your-path-to-mission-planner/Missionplanner.exe"`
4. (Optional) Create a script / bash alias to run the command, for example my bash_aliases has `alias mp='sg no-internet "mono ~/Documents/mission_planner/MissionPlanner.exe"'`

If you need to update the firmware on your vehicle then you will need to run mission planner with internet access for some reason. But, as you don't have to connect to the board mission planner won't run the new firmware check, so this isn't an issue.

Alternatively, you could disable internet access to your machine to use mission planner, but this doesn't work well for me as turning internet on and off to use mission planner is a huge nuisance.

Transistorsis, 2022-12-20:
This appears to be fixed in version 1.3.79 build 1.3.8375.24878.