ArduPilot/MissionPlanner #2763: Planner: add MAV_CMD_DO_SPRAYER (216) to list of available commands for Rovers/Boats

rmackay9, 2022-01-04:
Rover-4.1 (and maybe earlier) support the SPRAYER command (216)

Only param1 is consumed.  If it is 1 then the sprayer is turned on.  If it is zero the sprayer is turned off so maybe the column title for Param1 should be "off(0)/on(1)".

EosBandi, 2022-01-04:
There is this PR https://github.com/ArduPilot/MissionPlanner/pull/2761 which makes it possible for users to add any Mavlink commands to the dropdown list and store the added commands in the config.xml.

framosnt, 2022-01-10:
I was talk about this features feel days ago. I need this to use by Sprayer Agricultural Copter. Can this functionality make to Copters too ?

rmackay9, 2023-08-03:
Fixed long ago, thanks!

closed 2023-08-03 by rmackay9