ArduPilot/MissionPlanner #2782: Add Baro device id info to HWID page

rmackay9, 2022-01-27:
4.1 (and higher) includes BARO1/2/3_DEVID parameter so it would be nice to get these added to the HWID page as well if possible.
[image]

I noticed this as part of handling this support request about a broken baro. https://discuss.ardupilot.org/t/cubeblack-can-not-detect-its-baro/79045/18

This is the PR that added the device ids I think. https://github.com/ArduPilot/ardupilot/pull/14847

meee1, 2022-01-27:
this should come for free, as MP already has the baro code

meee1, 2022-01-27:
sorry i was wrong, the param names between that commit and now hav changed. updated as needed

rmackay9, 2022-01-27:
great, thanks!

rmackay9, 2023-08-03:
Baro devices appear in the HWID page now.  closing, txs!

closed 2023-08-03 by rmackay9