ArduPilot/MissionPlanner #2789: ADS-B server Information lost on exiting Mission Planner

timtuxworth, 2022-02-02:
#### Issue details
ADS-B server IP address/port on the Configuration tab is not persisted. If you go to Config (scroll down) check the ADSB checkbox and enter an IP address and port, it will work fine for the current session, displaying traffic on the map, but if you exit MP, it doesn't persist this information, so you have to re-enter ADS-B feed information every time you start Mission Planner.

#### Version
Mac OSX build 637 (About shows Version 1.0 (1))

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
I use planes, but this seems to happen regardless (even if I don't connect to a plane)

#### Hardware type
PixRacer R15

#### Logs
If there is a log that will show this please let me know what you need and I'll upload it.

meee1, 2022-02-07:
was this in a VM? or native?

meee1, 2022-02-07:
ive just added a patch that will save the config file more often. uptill not MP only saves the config on a clean exit, which may not be happening

timtuxworth, 2022-02-07:
> was this in a VM? or native?

It was native

timtuxworth, 2022-02-07:
> ive just added a patch that will save the config file more often. uptill not MP only saves the config on a clean exit, which may not be happening

Pretty sure it had exited cleanly, but Ill test this. Thamks Michael

timtuxworth, 2022-02-07:
Tested build #564 - it works! ADSB checkpoint and IP configuration is now persisted. ADSB traffic shows up as soon as the app is started.
MacBook Pro M1Pro OSX 12.2

meee1, 2022-02-07:
@timtuxworth thanks for the check

closed 2022-02-07 by meee1