ArduPilot/MissionPlanner #2799: After start MP disk usage 100%

aLDime, 2022-02-24:
#### Issue details
After start MP, within 15-20 minutes, the loading of disk "E" (the LOGs are stored there) is 100%

#### Version
Latest stable v1.3.76

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
N/A

#### Hardware type
N/A

#### Logs
N/A

#### Screen Shot
[image]

aLDime, 2022-03-05:
Maybe someone will say me why the MP scans 2TB hard drive completely all 20 minutes?

meee1, 2022-03-06:
got the help screen im MP, and turn on Console, restart MP,

and note what is happening.

aLDime, 2022-03-06:
Flight logs are stored on disk "E"

[image]
BT_DEVNODES_CHANGED 0
[image]

from Console:

If your error is about Microsoft.DirectX.DirectInput, please install the latest directx redist from here http://www.microsoft.com/en-us/download/details.aspx?id=35

Debug under mono    MONO_LOG_LEVEL=debug mono MissionPlanner.exe
To fix any filename case issues under mono use    export MONO_IOMAP=drive:case
for pinvoke      MONO_LOG_LEVEL=debug MONO_LOG_MASK=dll mono MissionPlanner.exe
Data Dir C:\ProgramData\Mission Planner\
Log Dir C:\Users\aLDime_E5\Documents\Mission Planner\logs
Running Dir C:\Program Files (x86)\Mission Planner\
User Data Dir C:\Users\aLDime_E5\Documents\Mission Planner\
INFO MissionPlanner.Program - ******************* Logging Configured *******************
INFO MissionPlanner.Program - C:\Program Files (x86)\Mission Planner\x64\libSkiaSharp
INFO MissionPlanner.Program - SkiaLoaded
4.0.0.0
Splash .ctor
IconFile
Splash.Show()
Debugger.IsAttached False
Application.DoEvents
Application.DoEvents
Setup GMaps 1
Setup GMaps 2
Setup GoogleMapProvider API
Setup Tracking.productName
Setup Tracking.primaryScreenBitsPerPixel
Setup Settings.Instance.UserAgent
Setup check gdal dir
Setup gdal
INFO GDAL.GDAL - GDAL static ctor
Setup proxy
Setup proxy
Setup CleanupFiles
INFO MissionPlanner.Program - 64bit os True, 64bit process True, OS Arch X64, OS Desc Microsoft Windows 10.0.19042 , FW Desc .NET Framework 4.8.4420.0
INFO MissionPlanner.Program - Runtime Version v4.0.30319
Application.Run(new MainV2())
INFO MissionPlanner.MainV2 - Mainv2 ctor
INFO MissionPlanner.MainV2 - Loading config
INFO MissionPlanner.MAVLinkInterface - Subscribed .ctor at offset 1515 in file:line:column <filename unknown>:0:0
 adsb_UpdatePlanePosition MainV2 System.EventHandler`1[MissionPlanner.Utilities.adsb+PointLatLngAltHdg]
INFO MissionPlanner.MAVLinkInterface - Subscribed .ctor at offset 1576 in file:line:column <filename unknown>:0:0
 <.ctor>b__96_0 MainV2 System.EventHandler`1[System.ValueTuple`2[System.String,MAVLink+MAV_COLLISION_THREAT_LEVEL]]
Background to Color [A=255, R=38, G=39, B=40]
Control Background to Color [A=255, R=67, G=68, B=69]
Text to Color [White]
TextBox Background to Color [A=255, R=67, G=68, B=69]
Button Text to Color [A=255, R=64, G=87, B=4]
Button Background top to Color [A=255, R=148, G=193, B=31]
Button Background bottom to Color [A=255, R=205, G=226, B=150]
ProgressBar Top to Color [A=255, R=102, G=139, B=26]
ProgressBar Bottom to Color [A=255, R=124, G=164, B=40]
ProgressBar Outline to Color [A=255, R=150, G=174, B=112]
BannerColor1 to Color [A=255, R=64, G=87, B=4]
BannerColor2 to Color [A=255, R=148, G=193, B=31]
Disabled Button to Color [A=150, R=43, G=58, B=3]
Button Mouseover to Color [A=73, R=43, G=58, B=3]
Button Mousedown to Color [A=73, R=43, G=58, B=3]
CurrentPPM Background to Color [Green]
Graph Chart Fill to Color [A=255, R=31, G=31, B=32]
Graph Pane Fill to Color [A=255, R=55, G=55, B=56]
Graph Legend Fill to Color [A=255, R=133, G=132, B=131]
Rich Text Box text to Color [WhiteSmoke]
BackStageView Button Area to Color [Black]
BSV Unselected Text to Color [WhiteSmoke]
[... 506 lines cut ...]
HUD 1 hz drawtime 4 gl True
HUD 1 hz drawtime 4 gl True
HUD 1 hz drawtime 3 gl True
HUD 1 hz drawtime 3 gl True
HUD 1 hz drawtime 4 gl True
HUD 1 hz drawtime 3 gl True
HUD 1 hz drawtime 94 gl True
DFLogBuffer-linecount: 2002288 time(ms): 226531,5253
HUD 1 hz drawtime 5 gl True
HUD 1 hz drawtime 2 gl True
HUD 1 hz drawtime 4 gl True
HUD 1 hz drawtime 3 gl True
HUD 1 hz drawtime 4 gl True
HUD 1 hz drawtime 4 gl True
HUD 1 hz drawtime 3 gl True
HUD 1 hz drawtime 3 gl True
HUD 1 hz drawtime 3 gl True
HUD 1 hz drawtime 3 gl True
HUD 1 hz drawtime 3 gl True
HUD 1 hz drawtime 4 gl True
HUD 1 hz drawtime 4 gl True
DFLogBuffer-linecount: 2308301 time(ms): 239002,4843
HUD 1 hz drawtime 3 gl True
HUD 1 hz drawtime 3 gl True
HUD 1 hz drawtime 2 gl True

meee1, 2022-03-06:
so from that i can tell you its reading dataflash logs
via the
DFLogBuffer-linecount: 1584177 time(ms): 184148,5942
lines.

mission planner will be scanning them to try sort them in to  the correct subdirectorys, and also genreate the jpg image preview of the flight path

meee1, 2022-03-06:
on a side note, your read io performance is poor, MP will only be reading the log, and writting if needed a jpg file, or a simple file move

aLDime, 2022-03-06:
Ok, I figured out what's going on, thanks for the tips! Cleaned out the "LOG's" folder, archived and deleted everything. Now loading MP is normal. But I think it's not normal, in principle, to manually clean all subfolders, even if by the search mask ..
Can I somehow automate the process in the future, for example, make the "Clear Local Logs" button in MP?

[MPConsoleClean!.txt](https://github.com/ArduPilot/MissionPlanner/files/8192060/MPConsoleClean.txt)

[image]
?

meee1, 2022-03-06:
by default MP only scans inside the bin files if no  *binfile*.jpg file is found. as to create the missing jpg file. so its more of a question as to why it was failing on those logs

aLDime, 2022-03-06:
> bin files if no `binfile.jpg`

Now everything is clear, I manually moved the "BIN" files, creating new folders with dates and descriptions .. It turns out my fault)) It's good that they explained this too! Issue is solved! I'm grateful a lot!

closed 2022-03-06 by aLDime