ArduPilot/MissionPlanner #2800: Mission Planner is unable to load firmware to external flash boards

andyp1per, 2022-02-24:
#### Issue details
ArduPilot 4.2 supports flight controllers running on external flash like the SPRacing H7 Extreme. Mission Planner is unable to load firmware to these boards.

[image]

#### Version
1.3.76 1.3.8029.15962

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
N/A

#### Hardware type
SPRacing H7 Extreme

#### Logs
N/A

meee1, 2022-02-25:
is there a reference example to this somewhere? i did know about this, but was asked to hold off a while.

meee1, 2022-02-25:
@andyp1per could you test the current beta MP?

meee1, 2022-02-25:
one query i do have, is why did the bootloader version not change when adding external flash support. seems kinda broken to me

andyp1per, 2022-04-12:
This is now fixed