ArduPilot/MissionPlanner #2818: Previously stored .grid files does not open in later versions of MP, added "laps" xml-parameter

atobiese, 2022-03-16:
#### Issue details
_Please describe the problem, or desired feature_
Hi!
Previously stored .grid files does not open in latest updated version of MP:
[image]

There is an issue with the xml definition, by which a new parameter is added:
< laps >200< /laps >

If this is not available in the previous files, it will not open. Previous versions will open if this is added. 
 
The error report is shown below:
[image]

Thanks and best regards

meee1, 2022-03-17:
can you include the grid file?

meee1, 2022-03-17:
what the error is saying is something is set to 0 in the grid file, and its not a valid value for that field.  just need to figure out what field it is

meee1, 2022-03-17:
at a guess the alt is 0, and should be >=5

atobiese, 2022-03-17:
[grid_removed_coordinates.grid.txt](https://github.com/ArduPilot/MissionPlanner/files/8283291/grid_removed_coordinates.grid.txt)

Thanks for the quick feedback, Michael. 
[image]

Enclosed is a test that the error message in my MP-install. Running latest version (updated today). It does open after adding the "laps" - parameter to the schema. 

yes, in this work flow, it seems as alt is indeed 0, but the default value from the waypoint view sets the grid UI. Ie, the alt parameter in that case is always 0 in the .grid files. 

Thanks again