ArduPilot/MissionPlanner #2820: Cannot download SITL firmware to start simulation

silentjet, 2022-03-18:
#### Issue details
While trying to download a SITL image for the plane in the simulation tab, the image download is failing.

Debian 11/amd64
Mono 6.12

#### Version
1.3.77 official

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[x] Plane
[  ] Rover
[  ] Sub

#### Airframe type
simu

#### Hardware type
simu

#### Logs
```
INFO MissionPlanner.ArduPilot.APFirmware - device: {
  "name": null,
  "description": null,
  "board": "",
  "hardwareid": ""
}
INFO MissionPlanner.Utilities.Download - https://firmware.ardupilot.org/Plane/stable/SITL_x86_64_linux_gnu/arduplane
INFO MissionPlanner.Utilities.Download - getFilefromNet(): System.Net.WebException: Error: TrustFailure (Authentication failed, see inner exception.) ---> System.Security.Authentication.AuthenticationException: Authentication failed, see inner exception. ---> Mono.Btls.MonoBtlsException: Ssl error:1000007d:SSL routines:OPENSSL_internal:CERTIFICATE_VERIFY_FAILED
  at /build/mono-6.12.0.122/external/boringssl/ssl/handshake_client.c:1132
  at Mono.Btls.MonoBtlsContext.ProcessHandshake () [0x00048] in <9c6e2cb7ddd8473fa420642ddcf7ce48>:0 
  at Mono.Net.Security.MobileAuthenticatedStream.ProcessHandshake (Mono.Net.Security.AsyncOperationStatus status, System.Boolean renegotiate) [0x000da] in <9c6e2cb7ddd8473fa420642ddcf7ce48>:0 
  at (wrapper remoting-invoke-with-check) Mono.Net.Security.MobileAuthenticatedStream.ProcessHandshake(Mono.Net.Security.AsyncOperationStatus,bool)
  at Mono.Net.Security.AsyncHandshakeRequest.Run (Mono.Net.Security.AsyncOperationStatus status) [0x00006] in <9c6e2cb7ddd8473fa420642ddcf7ce48>:0 
  at Mono.Net.Security.AsyncProtocolRequest.ProcessOperation (System.Threading.CancellationToken cancellationToken) [0x000fc] in <9c6e2cb7ddd8473fa420642ddcf7ce48>:0 
   --- End of inner exception stack trace ---
  at Mono.Net.Security.MobileAuthenticatedStream.ProcessAuthentication (System.Boolean runSynchronously, Mono.Net.Security.MonoSslAuthenticationOptions options, System.Threading.CancellationToken cancellationToken) [0x00262] in <9c6e2cb7ddd8473fa420642ddcf7ce48>:0 
  at Mono.Net.Security.MonoTlsStream.CreateStream (System.Net.WebConnectionTunnel tunnel, System.Threading.CancellationToken cancellationToken) [0x0016a] in <9c6e2cb7ddd8473fa420642ddcf7ce48>:0 
  at System.Net.WebConnection.CreateStream (System.Net.WebOperation operation, System.Boolean reused, System.Threading.CancellationToken cancellationToken) [0x001ba] in <9c6e2cb7ddd8473fa420642ddcf7ce48>:0 
   --- End of inner exception stack trace ---
  at System.Net.WebConnection.CreateStream (System.Net.WebOperation operation, System.Boolean reused, System.Threading.CancellationToken cancellationToken) [0x0021a] in <9c6e2cb7ddd8473fa420642ddcf7ce48>:0 
  at System.Net.WebConnection.InitConnection (System.Net.WebOperation operation, System.Threading.CancellationToken cancellationToken) [0x00141] in <9c6e2cb7ddd8473fa420642ddcf7ce48>:0 
  at System.Net.WebOperation.Run () [0x0009a] in <9c6e2cb7ddd8473fa420642ddcf7ce48>:0 
  at System.Net.WebCompletionSource`1[T].WaitForCompletion () [0x00094] in <9c6e2cb7ddd8473fa420642ddcf7ce48>:0 
  at System.Net.HttpWebRequest.RunWithTimeoutWorker[T] (System.Threading.Tasks.Task`1[TResult] workerTask, System.Int32 timeout, System.Action abort, System.Func`1[TResult] aborted, System.Threading.CancellationTokenSource cts) [0x000f8] in <9c6e2cb7ddd8473fa420642ddcf7ce48>:0 
  at System.Net.HttpWebRequest.GetResponse () [0x00016] in <9c6e2cb7ddd8473fa420642ddcf7ce48>:0 
  at MissionPlanner.Utilities.Download.getFilefromNet (System.String url, System.String saveto, System.Action`2[T1,T2] status) [0x0006f] in <78bc2f1b850e4a7a8aab0cc60aa4bdfa>:0 
CustomMessageBox.Show
CustomMessageBox thread calling Base Thread
CustomMessageBox thread running Base Thread
```

meee1, 2022-03-18:
update your mono version.

this is a certificate issue, most likerly lets encrypt issue

meee1, 2022-03-18:
try a 
sudo apt-get install ca-certificates-mono

silentjet, 2022-04-08:
After these steps I do see some certificates were imported/added, however now 1.3.77 could not start at all, while 1.3.76 - still works fine.
[image]

meee1, 2022-04-09:
are you in a country that doesnt modify ssl certs on the fly?

silentjet, 2022-04-09:
You've surprised me with such a question. I assume nobody's modifying my internet traffic including certificates... I'm in EU...

silentjet, 2022-07-30:
All right. It seems the issue occurs only under one circumstance if the current directory is MissionPlanner dir. If I'm running having current dir as HOME and using a full path to exe as an argument to the mono command, then MP do not stuck at that sync https actions.

silentjet, 2022-11-06:
@meee1 
All right. Now it's clear. So the issue is about "Let's Encrypt" who changed the root certificates and chain of trust, and the old one expired (Sep'21) and blah-blah-blah...
And since autotest.ardupilot.org is using the certificate issued/chained with Let's Encrypt's one, the Debian is trapping into an issue that the verification chain is going through the old Mozilla cert path and reaching expired cert.

The verification step is simple:
`csharp -e 'new System.Net.WebClient ().DownloadString ("https://autotest.ardupilot.org/LogMessages/Copter/LogMessages.xml.xz")'`

Some technical details are in here: https://community.letsencrypt.org/t/r3-intermediate-certificate-has-expired-how-to-fix-this-on-debian-with-certbot/160896/14

The fix that worked for me is to commend out certificate "mozilla/DST_Root_CA_X3.crt" from /etc/ca-certificates.conf , then update certificates with `update-ca-certificates` and now MP can start.

Obviously, that is not really a perfect or even good way, but to be honest I do not know how to change that. Maybe some "trusted" certificates can be delivered altogether with MP distribution?