ArduPilot/MissionPlanner #2833: Extended tuning page: roll/pitch FLTD and FLTE values are swapped

RickReeser, 2022-04-08:
#### Issue details
The FLTD and FLTE fields for rate roll modify the parameters `ATC_RAT_PIT_FLTD` and `_FLTE`. The fields for pitch modify the respective roll parameters.

[image]

[image]

#### Version
1.3.77

RickReeser, 2022-04-08:
I tried regenerating the param info, no change.

meee1, 2022-04-10:
77 fixs this,  im not sure what your seeing

rmackay9, 2022-04-10:
I had noticed this issue as well (and though I raised an issue for it but can't find it now) but in any case, as MichaelO says, it's fixed in the latest beta.  So if you go to the Help screen and push "Check for BETA Updates" it should install 1.3.77 and the problem will be resolved.  The next stable release will include the fix as well I'm sure.
[image]