ArduPilot/MissionPlanner #2866: Fence Mission Items Not Supported Please Use Legacy Geofence

flybrianfly, 2022-05-23:
#### Issue details
I'm trying to upload a polygon exclusion geofence to my [company's flight stack](https://github.com/bolderflight/spaaro). We are reporting the following capabilities:
   * MAV_PROTOCOL_CAPABILITY_PARAM_FLOAT
   * MAV_PROTOCOL_CAPABILITY_MISSION_INT
   * MAV_PROTOCOL_CAPABILITY_MAVLINK2
   * MAV_PROTOCOL_CAPABILITY_MISSION_FENCE
   * MAV_PROTOCOL_CAPABILITY_MISSION_RALLY

And our autopilot is reported as MAV_AUTOPILOT_GENERIC.

I'm able to upload fences with QGC. In Mission Planner, when I write the fence, I get this exception:
Fence Mission Items Not Supported Please Use Legacy Geofence
[image]

#### Version
1.3.77 and 1.3.77 with Beta updates.

meee1, 2022-05-24:
what you are sending looks ok, can you provide a tlog?

flybrianfly, 2022-05-24:
@meee1, attached is a zip of the tlog.
[2022-05-23 19-22-12.zip](https://github.com/ArduPilot/MissionPlanner/files/8758900/2022-05-23.19-22-12.zip)

The mavlink_autopilot_version_t message is sent in response to a mav_cmd_request_message_t being received.

flybrianfly, 2022-05-24:
Shoot, I think that's the disconnect, I'm using the command microservice for the GCS to send the MAV_CMD_REQUEST_AUTOPILOT_CAPABILITIES, which I don't see the autopilot getting, thanks for the hint.

meee1, 2022-05-24:
MP uses mavlink_autopilot_version_request_t

flybrianfly, 2022-05-24:
Thanks! That fixes it.

closed 2022-05-24 by flybrianfly