ArduPilot/MissionPlanner #2868: Plan: right-mouse-button menu options different in Japanese vs English

rmackay9, 2022-05-25:
For some reason the number of items visible on the Plan screen's right-mouse-button menu are different in English vs Japanese.  I wonder if perhaps the menu builders skips items that don't have a local translation available?  I think if a local translation isn't available it should just use the english word
[image]

meee1, 2022-05-25:
is this the exact same install?

meee1, 2022-05-25:
perfect, thanks.  the missing items are plugins, being on the same install means its something about the plugins, not that the plugins are missing