ArduPilot/MissionPlanner #2881: Facemap is awesome! a few thoughts re dev

aritchie-usgs, 2022-06-09:
#### Issue details
_FaceMap is awesome but increase flexibility to improve flight planning_

Facemap is sooo coool! VERY helpful for me to plan oblique missions with a crewed aircraft - closest I found to actually getting me flight lines that would work - BUT:
*maximum angle is too low (45°) - ideally up to something much more oblique, say 70°
*max number of steps is low - would be nice to increase by 10x
*max elevation is too low, would be nice to increase by 10x
*max step height is too low, would be nice to increase by 10x

With those settings I can simulate a slope and calculate locations for flight lines. Would be even better if you could add groups of faces of different slopes.

Also stepping from all sides is weird, when I'm trying to plan mapping a canyon wall - I have to extend the polygon bounds way outside of the canyon so I can ignore those parts. Not sure how you could make it only go from one side of the poly though.

#### Version
_1.3.77 build 1.3.8110.38294 Windoze_

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_NA_

#### Hardware type
_NA_

#### Logs
_None_

meee1, 2022-06-09:
could you list some sample values you would like to use? just so i change the values enough to not limit you

aritchie-usgs, 2022-06-09:
**Face Map/grid options/distance from face** 0 - 1000m
---------------------------
**Face Map/Simple/Angle (deg)** 90- 30°
**Face Map/Simple/Number of Benches** 0-20
**Face Map/Simple/Camera pitch (deg)** 0-70 (could you just take it all the way to 90?)
---------------------------
**other** - any way to toggle an option to map the benches too? I realize those camera lines would be unevenly spaced, but that could help in planning for mesas or sets of marine terraces.

The most important one for me is **Face Map/grid options/distance from face** - I rarely fly higher than 1000m so I think this is adequate.

Going through all the **Face Map/Simple** params in order with comments:
**Bench height (m)** - this isn't limiting me - I can put in at least 1000m (didn't actually realize this)
**Angle (deg)** - 45° isn't bad, but 30° is ~ minimum angle of repose for unconsolidated soil, so might be good for a lower bound.
**Berm depth (m)** - not limiting (same as bench ht I think)
 **Number of benches** - I'm re-thinking this now that I realized bench and berm are unlimited - especially since camera lines don't cover the berm area. - maybe 20

**Camera pitch (deg)** - I realized that you are counting angle from horiz (horiz=0°) and I was counting angle from vertical - but I think going further to like 70° would be good. Most cams have a FOV that can handle the 70-90° case with vertical line planning.

meee1, 2022-06-09:
ok, i think most of those changes will already be in with the mod i did yesterday, please try beta MP