ArduPilot/MissionPlanner #2885: Plan: ATTITUDE_TIME climb rate field reset to zero

rmackay9, 2022-06-16:
Similar to https://github.com/ArduPilot/MissionPlanner/issues/2871 the Plan screen resets the ATTITUDE_TIME message's "ClimbRate" field to zero after it is written and then read back from the autopilot.
[image]

I suspect we will find the same thing happens to the DO_GIMBAL_MANAGER_PITCHYAW command once PR https://github.com/ArduPilot/MissionPlanner/pull/2882 is merged.

meee1, 2022-06-16:
https://github.com/ArduPilot/MissionPlanner/blob/a6c167aa0e52478d5f2d866ba3690bd2b889e825/ExtLibs/ArduPilot/Mavlink/MAVLinkInterface.cs#L3819
At first glance I set all fields, and Rx all fields.  So is the problem a scaling one?

rmackay9, 2022-06-16:
@meee1,

OK, thanks.  I must say that it acts a little differently than it did a week ago.  A week ago if I input a number into the ClimbRate field I think it would always be changed to 128 (on the autopilot side) and then when I retrieved it back in MP I would get 1.28e-7 (or something like that).  Now it always comes back as zero.

I guess maybe the MP side is fixed and I should go back and check the flight code side and be sure we don't have a mistake there..

rmackay9, 2022-06-17:
@meee1, 

I re-tested with latest beta MP and it (MP that is) seems to be sending very large values for the climb rate field so I guess you're right that it's scaling.  So I input "5" and AP receives 5e+7.  Somehow within AP that's then getting zapped back to zero.  I could perhaps fix AP so it can store big values but the core issue is the big climb rate which surely the user doesn't want.
[image]

rmackay9, 2022-06-23:
Txs for that fix.  I've re-tested the latest beta and it doesn't seem to fix the problem which is odd.  The do-gimbal-manager-pitchyaw does work though.  I'll check the flight code more..