ArduPilot/MissionPlanner #2887: Incorrect voltage readout on HUD

CallanDaniel, 2022-06-20:
#### Issue details
_There is a possible bug in the displayed voltage on the HUD - a 1 is appended to the front of the actual reading:_
[image]

#### Version
_1.3.77.1 build 1.3.8206.14892_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ X ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
_Quadplane_

#### Hardware type
_Tested on Cube Orange, Pixhawk 4 and SITL_

meee1, 2022-06-20:
can you provide a tlog?

meee1, 2022-06-20:
actualy, not needed, i tink i know why

CallanDaniel, 2022-06-20:
Cool, thanks. Shout if you need a tlog @meee1

closed 2022-06-20 by CallanDaniel

CallanDaniel, 2022-06-20:
Built locally and it worked - how long until this build will be available via the "Beta Update" button? Sorry, I know this is a silly question, but this version has a fix that allows us to use Silvus radios over UDP - and I'm not quite at the stage of compiling to an installation file I can send to clients... @mee