ArduPilot/MissionPlanner #2912: DO_JUMP not displaying correctly on map

tridge, 2022-07-20:
using DO_JUMP is normal in most fixed wing missions, but MissionPlanner doesn't display it correctly
here is a MissionPlanner display of a typical quadplane mission:
[image]
here is the same mission in MAVProxy:
[image]
notice the correct line for the segment following the DO_JUMP, and the correct display of the DO_LAND_START
the mission file for this example is attached
Thanks!
[wp.txt](https://github.com/ArduPilot/MissionPlanner/files/9154680/wp.txt)

rmackay9, 2022-07-20:
Ah, so if there is a do-jump then the wp before the do-jump needs to display two lines.  One to the command index from the do-jump command and one to the command after the do-jump.

tridge, 2022-07-21:
> Ah, so if there is a do-jump then the wp before the do-jump needs to display two lines. One to the command index from within the do-jump command and one to the command after the do-jump.

no, I think we should only display the "jump is taken" case. That is what MAVProxy has always done and it has worked very well and is very easy to understand