ArduPilot/MissionPlanner #2934: Support Holybro 6X upload via MP

rmackay9, 2022-08-30:
It seems that the Holybro 6X boards aren't recognised by MP as an autopilot that it can upload to.

tridge, 2022-08-30:
The issue only happens on new boards that are running the px4 firmware that is shipped on the board. MissionPlanner never gets to the step of asking the users to unplug and press OK, it just fails with:
[image]
Once ArduPilot is installed on the board then everything works fine. Also note that load custom firmware does work when it is running px4 firmware.
this is in the MissionPlanner log file:
```
  "name": "COM59",
  "description": "USB Serial Device",
  "board": "PX4 FMU v6X.x",
  "hardwareid": "USB\\VID_3185&PID_0035&REV_0101"
} (D:\a\MissionPlanner\MissionPlanner\ExtLibs\ArduPilot\APFirmware.cs:134) [Base Thread]
2022-08-31 09:20:59,700  INFO MissionPlanner.ArduPilot.APFirmware - device: {
  "name": "COM1",
  "description": "Communications Port",
  "board": null,
  "hardwareid": "ACPI\\VEN_PNP&DEV_0501"
} (D:\a\MissionPlanner\MissionPlanner\ExtLibs\ArduPilot\APFirmware.cs:134) [Base Thread]
2022-08-31 09:20:59,709  INFO MissionPlanner.ArduPilot.APFirmware - device: {
  "name": "COM59",
  "description": "USB Serial Device",
  "board": "PX4 FMU v6X.x",
  "hardwareid": "USB\\VID_3185&PID_0035&REV_0101"
```
full log file attached
[MissionPlanner.log](https://github.com/ArduPilot/MissionPlanner/files/9457168/MissionPlanner.log)

meee1, 2022-08-30:
can we add this vid/pid to thefw  manifest?

tridge, 2022-08-30:
@meee1 yes, I think we can, I'll have a go

meee1, 2022-08-30:
@tridge and the bootloader string as well  "PX4 FMU v6X.x"

meee1, 2022-08-30:
also, a possible workaround
open the FW screen
then disconnect and reconnect. the FW screen should attempt to hold it in bootloader mode

tridge, 2022-08-30:
@meee1 this might fix it:
https://github.com/ArduPilot/ardupilot/pull/21581
note the duplicated USB IDs

meee1, 2022-08-30:
ok, someone with the HW might need to test the specific outcome, one the manifest refresh's

tridge, 2022-08-30:
@meee1 will do

meee1, 2022-08-31:
i have a feeling the duplicate vid/pid might cause issues. but adding a new change to hopefully handle that

tridge, 2022-08-31:
@meee1 I've tested with the updated manifest and it works, thanks!

closed 2022-08-31 by tridge