ArduPilot/MissionPlanner #2939: Stuck, "Getting Params 250"

rmackay9, 2022-09-07:
For some reason when running Mission Planner connected via HereLink to the vehicle, MP got stuck displaying the message, "Getting Params 250".  It never seemed to give-up and the pop-up window was modal so the only solution was to restart MP.

[image]

meee1, 2022-09-07:
what device is sysid 250?

rmackay9, 2022-09-07:
@meee1,

I'm not sure but if the "250" is a component ID then this is what I see.  The "vehicle 10" is probably the Gremsy gTuneDesktop application.
[image]

So this may be a real edge-case that users will almost never see.  I was just surprised that MP would get stuck like this though because I guess that means a timeout is not being implemented properly somewhere right?

rmackay9, 2023-03-10:
I think this is fixed now

closed 2023-03-10 by rmackay9