ArduPilot/MissionPlanner #2961: Data: Payload Control should use limits from MNT1_

rmackay9, 2022-10-05:
The Payload Control page can be used to control a gimbal mount's target angles using slides.  It seems that the limits for all 3 axis are -4500 to +4500 but these should really match the limits in the MNT1_ parameters.  Note that both the names have changed (see below) and the scaling has changed from centi-degrees to degrees. 

- MNT**1**_PITCH_MIN, MNT1_PITCH_MAX (in 4.2 was MNT_ANGMIN_TIL, MNT_ANGMAX_TIL in centi-degrees)
- MNT**1**_ROLL_MIN, MNT1_ROLL_MAX (in 4.2 was MNT_ANGMIN_ROL, MNT_ANGMAX_ROL in centi-degrees)
- MNT**1**_YAW_MIN, MNT1_YAW_MAX (in 4.2 was MNT_ANGMIN_PAN, MNT_ANGMAX_PAN in centi-degrees)

By the way, the Payload Control page sends MOUNT_CONTROL and MOUNT_CONFIGURE messages.  There's no need for it to send MOUNT which should probably be updated to use

This is somewhat related to this enhancement request https://github.com/ArduPilot/MissionPlanner/issues/2906

rmackay9, 2022-10-05:
I've re-tested with the latest beta and the increased ranges (-90 to +90 for pitch and roll, -180 to +180 for yaw) is working great, thanks!

muramura, 2022-10-12:
I would add a selection for MNT1 and MNT2.
This selection determines the group.

Making Mount
[image]

rmackay9, 2022-10-12:
@muramura that MNT1/MNT2 drop-down looks good to me.  I'm not qualified to review MP code though so I'll leave it to @meee1 to make the decisions.