ArduPilot/MissionPlanner #2981: 16:9 OSD

stevegut78, 2022-10-15:
#### Issue details
With the release of fpv.wtf and MSP OSD, is it possible to have the option to change the OSD screen in MP to reflect the screen size of HD? At the moment, some elements are off-screen in MP and require manual X/Y coordinates. It would be nice to be able to see the full OSD layout in MP when organizing the OSD elements.

#### Version
1.3.77.1

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
Wing

#### Hardware type
Matek F405Wing

#### Logs
_Please provide a link to any relevant logs that show the issue_

maribeiro, 2022-12-09:
I would also appreciate if we could have this working.

robustini, 2023-02-22:
Hi @meee1, please I am also waiting for this integration, it would be useful for the Planner to show the HD matrix to place the elements correctly, based on the value of the OSDx_TXT_RES parameter.
I always use the 50x18 matrix and it is somewhat inconvenient to arrange things while wearing glasses.
Currently the Planner shows only the 30x16 resolution.

shannonbaker, 2023-07-06:
Given the proliferation of HD FPV/OSD solutions, this fix should ideally be prioritised.

Duese123, 2023-11-23:
do there have to be something special? iam not seeing this... 
[image]

robustini, 2023-11-24:
> do there have to be something special? iam not seeing this...

In fact there isn't, in order to be able to, for example, create a layout like this i'm going crazy working blindly, i can't use the Planner to preview but the output video.
Betaflight, INAV, Rotorflight and the like, on the other hand, have the graphical preview with a correct matrix in their configurators.

[image]

robertlong13, 2023-11-24:
Update to the latest beta

robustini, 2023-11-24:
> Update to the latest beta

Thanks!
Finally, very well, this issue can be closed.

[image]