ArduPilot/MissionPlanner #2986: FFT display fails on some logs and is inaccurate on others with pre/post concurrent FFTs

Hwurzburg, 2022-10-24:
This log fails to display at all: https://www.dropbox.com/s/862n77ezp9tial2/FFT%20filter%20log.bin?dl=0
but does so in mavfft_isb.py

This log displays incorrect Gyro FFT levels: https://www.dropbox.com/s/zqqvgqili046u5a/FFT%20filter%20log2.bin?dl=0

correct display
[image]
MP display
[image]

meee1, 2022-10-27:
@Hwurzburg any idea why there are 4 imus in the log that doesnt load?

Hwurzburg, 2022-10-27:
the option to do pre/post filtering concurrently labels the first ACC pre samples as ACC(0), the second as ACC(1), the first post filter as ACC(3) and the second as ACC(4)....same for gyros pre and post