ArduPilot/MissionPlanner #3008: Missionplanner - calculate height with 5cm accuracy

soleongmbh, 2022-11-24:
#### Issue details
I miss the possibility to calculate the height with 5cm accuracy for missions? Currently, the height is rounded to 1 metre. In addition, we would also like to be able to enter the default height to the centimetre when planning (currently only in metres). Is this possible?
[image]

[image]

#### Version
Mission Planner 1.3.77 build 1.3.8110.38294
ArduCopter V4.2.3

#### Platform
[  ] All
[  ] AntennaTracker
[X  ] Copter
[  ] Plane
[  ] Rover
[  ] Sub

#### Airframe type
all copter frames

#### Hardware type
Cube orange

#### Logs
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2022-11-25:
I did a quick test and I think the Plan screen's altitude field accepts decimal points (e.g. 10.25m).  I wrote and re-read the mission and the decimal points remained and the vehicle correctly flew to the specified altitude.

soleongmbh, 2022-11-25:
@rmackay9 Thanks for the test! Even if we import missions with decimal points, it works without any problems.  The decimal points must be enabled for the Plan screen's "Default Alt" and the "Auto WP" >> "Simple Grid"'s and "Survey Grid"'s "Altitude".

soleongmbh, 2023-03-31:
I have found another problem. If the planning is done with "Absolute", then the height is calculated exactly in cm. However, if a WP point is moved, then the decimal places are lost.
This is very dangerous when flying at a low altitude.
With "Relative" the decimal places are also lost.
[image]

Clicked in the map (Absolute) without moving
Clicked in map (Absolute) and moved
Clicked in the map (Relative) without moving

Can someone please correct this urgently?

closed 2023-05-08 by soleongmbh