ArduPilot/MissionPlanner #3016: firmware installs failing

tridge, 2022-12-06:
all firmware installs are failing. I checked in apache2 logs, and the apj is being downloaded successfully, and is the right size
this is in MP log:
```
2022-12-06 11:54:21,099 DEBUG MissionPlanner.Program - FirstChanceException in: System (C:\Users\mich1\Desktop\CubePilot\MissionPlanner\Program.cs:524) [37]
System.ObjectDisposedException: Cannot access a disposed object.
Object name: 'System.Net.Sockets.Socket'.
   at System.Net.Sockets.Socket.EndReceiveFrom(IAsyncResult asyncResult, EndPoint& endPoint)
2022-12-06 11:54:21,101 DEBUG MissionPlanner.Program - FirstChanceException in: System (C:\Users\mich1\Desktop\CubePilot\MissionPlanner\Program.cs:524) [37]
System.ObjectDisposedException: Cannot access a disposed object.
Object name: 'System.Net.Sockets.Socket'.
   at System.Net.Sockets.Socket.EndReceiveFrom(IAsyncResult asyncResult, EndPoint& endPoint)
   at System.Net.Sockets.UdpClient.EndReceive(IAsyncResult asyncResult, IPEndPoint& remoteEP)
   at System.Net.Sockets.UdpClient.<ReceiveAsync>b__64_1(IAsyncResult ar)
   at System.Threading.Tasks.TaskFactory`1.FromAsyncCoreLogic(IAsyncResult iar, Func`2 endFunction, Action`1 endAction, Task`1 promise, Boolean requiresSynchronization)
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
2022-12-06 11:54:21,099 DEBUG MissionPlanner.Program - FirstChanceException in: System (C:\Users\mich1\Desktop\CubePilot\MissionPlanner\Program.cs:524) [36]
System.ObjectDisposedException: Cannot access a disposed object.
Object name: 'System.Net.Sockets.Socket'.
   at System.Net.Sockets.Socket.EndReceiveFrom(IAsyncResult asyncResult, EndPoint& endPoint)
2022-12-06 11:54:21,102 DEBUG MissionPlanner.Program - FirstChanceException in: System (C:\Users\mich1\Desktop\CubePilot\MissionPlanner\Program.cs:524) [36]
System.ObjectDisposedException: Cannot access a disposed object.
Object name: 'System.Net.Sockets.Socket'.
   at System.Net.Sockets.Socket.EndReceiveFrom(IAsyncResult asyncResult, EndPoint& endPoint)
   at System.Net.Sockets.UdpClient.EndReceive(IAsyncResult asyncResult, IPEndPoint& remoteEP)
   at System.Net.Sockets.UdpClient.<ReceiveAsync>b__64_1(IAsyncResult ar)
   at System.Threading.Tasks.TaskFactory`1.FromAsyncCoreLogic(IAsyncResult iar, Func`2 endFunction, Action`1 endAction, Task`1 promise, Boolean requiresSynchronization)
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
2022-12-06 11:54:21,099 DEBUG MissionPlanner.Program - FirstChanceException in: System (C:\Users\mich1\Desktop\CubePilot\MissionPlanner\Program.cs:524) [35]
System.ObjectDisposedException: Cannot access a disposed object.
Object name: 'System.Net.Sockets.Socket'.
   at System.Net.Sockets.Socket.EndReceiveFrom(IAsyncResult asyncResult, EndPoint& endPoint)
2022-12-06 11:54:21,104 DEBUG MissionPlanner.Program - FirstChanceException in: System (C:\Users\mich1\Desktop\CubePilot\MissionPlanner\Program.cs:524) [35]
System.ObjectDisposedException: Cannot access a disposed object.
Object name: 'System.Net.Sockets.Socket'.
   at System.Net.Sockets.Socket.EndReceiveFrom(IAsyncResult asyncResult, EndPoint& endPoint)
   at System.Net.Sockets.UdpClient.EndReceive(IAsyncResult asyncResult, IPEndPoint& remoteEP)
   at System.Net.Sockets.UdpClient.<ReceiveAsync>b__64_1(IAsyncResult ar)
   at System.Threading.Tasks.TaskFactory`1.FromAsyncCoreLogic(IAsyncResult iar, Func`2 endFunction, Action`1 endAction, Task`1 promise, Boolean requiresSynchronization)
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()

```

rmackay9, 2022-12-06:
Here's what appears to the user
[image]

[This is the discussion where the issue was first noticed](https://discuss.ardupilot.org/t/copter-4-3-1-released/93929)

rmackay9, 2022-12-06:
This has been resolved with the latest beta

closed 2022-12-06 by rmackay9