ArduPilot/MissionPlanner #3022: Simulation: running wrong version of stable Copter, beta does not work

rmackay9, 2022-12-10:
How to reproduce:

- Open SIMULATION page
  - check Wipe checkbox (maybe not necessary)
  - click on Multirotor icon
  - when asked, select "stable"
  - check the version in the top left of the window <-- "ArduCopter V4.2.3 (6a694642)" appears
[image]

Repeat the above but after clikcing Multirotor, select "Latest(Dev)" and we see this failure:
[image]
[image]

Surely the problem is related to PR https://github.com/ArduPilot/ardupilot/pull/22356

tridge, 2022-12-10:
I've fixed the download script so it now does get 4.3 for stable, it gets the Copter-4.3 branch, which is currently 4.3.2-rc1

rmackay9, 2022-12-12:
@tridge, Txs for that.  It seems that now choosing "stable" causes 4.3.2-rc1 to be loaded... so the beta is being used when the user has asked for stable.

rmackay9, 2023-03-17:
I think this has been resolved now so closing.

closed 2023-03-17 by rmackay9