ArduPilot/MissionPlanner #3029: NAMED_VALUE_FLOAT always displays a value of zero in Quick tab

jeff567, 2022-12-20:
#### Issue details
NAMED_VALUE_FLOAT always displays a value of zero in the quick view tab and the tuning plot; however viewing the message with Mavlink Inspector shows that the correct value is being received by Mission Planner.  The correct value is also logged on the autopilot.

NAMED_VALUE_FLOAT is being sent from a companion computer using pymavlink.

#### Version
Initially tried with an older version of MP, but then tried with the latest version (1.3.79), with the same result.

[image]

After searching, I found a similar issue in the following forum post, but it is unresolved:
https://discuss.ardupilot.org/t/custom-field-value-always-0/71248

meee1, 2022-12-20:
It doesn't show as it's not from the sysid/compid you are viewing   and is technically by design.

jeff567, 2022-12-20:
Ok, thanks.  So in that case, it's not possible to view a live readout of a named float coming from an onboard computer?  It would have to originate from the autopilot itself (either through the firmware or a lua script)?

meee1, 2022-12-20:
Yes that was the design intent.  Ie this prevents the case of multiple compids sending the same message and things getting wierd. Ie affects all messages not just named float.

jeff567, 2022-12-20:
Understood, thank you for clarifying... I will close the issue.

closed 2022-12-20 by jeff567