ArduPilot/MissionPlanner #3031: multi-line @Bitmask parameters

tridge, 2022-12-29:
we are switching to multi-line @Bitmask parameter markup. 
See first example here:
https://github.com/ArduPilot/ardupilot/pull/22505
the reason we are doing this is both to make the cpp clearer, and because we've noticed the MissionPlanner display of long @Description blocks is unreadable, whereas long @Bitmask fields is usable.
We made this PR to try and test if MissionPlanner would pick up the changes:
https://github.com/ArduPilot/ardupilot/pull/22519
the test seems to show that MissionPlanner param regen is not seeing the changes. I've been trying to work out why.

What would we need to do in order for MissionPlanner to not use its own parser and instead use the pre-parsed xml we autobuild?

peterbarker, 2022-12-29:
It's also `@Values` - not used in master yet.

meee1, 2022-12-29:
MP uses
https://autotest.ardupilot.org/Parameters/
as the primary source, and falls back to the old method if nothing is found
ie blimp would fall back atm... fixing atm
so i need to address any display issues as display issues

tridge, 2022-12-29:
@meee1 what do I look for in MissionPlanner.log to see if it is falling back to the internal parser? Maybe we could remove the internal parser now so issues are clearer?

meee1, 2022-12-29:
so pdef data is cached for 7 days, so you need to clean the cache in c:\programfiles\missionplanner\*.pdef.*

tridge, 2022-12-29:
I deleted the pdef files and did the control-F param regen. The files re-appeared, but it is still the old data
[image]

meee1, 2022-12-29:
did you remove the .gz and xml?

tridge, 2022-12-29:
@meee1 ahh, no I didn't remove the .gz, thanks!
with those removed and another regen I get the right bitmask:
[image]
can we auto-remove all the cache files when a user presses the regen button?

closed 2023-03-10 by tridge