ArduPilot/MissionPlanner #3040: MissionPlanner is stuck at Getting Params after trying to connect to RFD900x

abhisheeekkk, 2023-01-12:
#### Issue details
Hi Guys, I have flashed and tried with latest ardupilot firmware of plane and copter in Cube Orange+, I have connected one RFD900x with Cube with TELEMETRY 1 Port, and given 5V power supply from external source, another RFD900x is connected to the PC with FTDI. RFD900x is also flashed with latest firmware(3.57) and hardware is V2.0 .
I have checked all the TX,RX Cables everything is connected fine.

When I open MissionPlanner and go to Optional Hardware -> SiK Radio and then I click on the load settings I am able to see all the parameters of the remote and local RFD900x device as shown in the below image.

[image]

After this when I try to connect COM6, with the baud rate 115200. I get connected but then while getting the parameters Mission Planner is stuck at getting params as shown below, although I am able to see HUD Parameters changing like, altitude, yaw, vertical speed, etc.

[image]

#### Things that I tried

I have tried these multiple things for the resolution of these issue but no luck :

Changed baud rate to 115200 and air speed to 200.
Changed baud rate to 57600and air speed to 64.
Changed baud rate to 9600 and air speed to 64.

I was also changing parameters like **SERIAL1_BAUD** to 115, 57 and 9, **SERIAL1_PROTOCOL** is 2 (MAVLink2), **SERIAL1_OPTIONS** is set to 0.

I tried changing Mavlink to standard and low latency both.
I tried changing MAX WINDOW to 131,80 and 20.
RTSCTS and ECC is untick.

I also tried interchanging both the RFD900x but still no luck.

#### Version
MissionPlanner 1.3.79 as you can see in above image.

#### Platform
[  ] All
[  ] AntennaTracker
[-] Copter
[-] Plane
[  ] Rover
[  ] Sub

I have tried with copter and plane firmware.

#### Airframe type
QuadCopter and Quadplane.

#### Hardware type
Cube Orange+

#### Logs
I am uploading the log file here.
[mission planner logs.txt](https://github.com/ArduPilot/MissionPlanner/files/10400479/mission.planner.logs.txt)

Please provide your possible resolution for this, Thanks.

tridge, 2023-01-17:
see #3044

Mirghani094, 2023-02-01:
If it took long time see #3044  
However, if it didn't start getting the parameters at all. Check your RFD connection. The parameters are requested from the GCS (Command sent from GCS to Autopilot), they are not send by default to the GCS.
If the TX cable from the GCS or the RX cable on the AP is not connected this issue will happen.