ArduPilot/MissionPlanner #3044: handling of packet loss in FTP - SIM_UART_LOSS

tridge, 2023-01-15:
Users have reported that MissionPlanner takes a very long time (several minutes) to download parameters on SiK radios. I looked into this yesterday on one users copter and found that the issue is how MissionPlanner handles gaps from the burst read in a FTP download.
To make this easy to reproduce I have added a new parameter in master SIM_UART_LOSS which is the percentage of lost bytes on all UARTs in SITL. Setting it to 0.05 means losing 1 byte every 2000 bytes, which means less than 1/10 of the burst read replies are lost. That is a reasonable simulation a common radio link.
Using SIM_UART_LOSS with MAVProxy the parameter ftp download of copter master takes under 4 seconds. With MissionPlanner SITL and the same setting it can take many minutes which is frustrating for users.
Some problems with the MissionPlanner approach:
 - the answer.Position != ftphead.offset check means that when one burst read reply packet is lost at the start of a long sequence that all the other replies after that are discarded. The FTP server in ArduPilot keeps sending the packets after this lost packet
 - the loss of a single packet results in a retry, but the retry also does a burst read, which means it asks for the whole file again, starting at the first lost packet offset. This just overloads the link more as the file is still being sent, resulting in more packet loss
 - after a number of retries the FTP is stopped and single value param reads are done at a very slow pace, often pausing for long periods

The MAVProxy approach is to maintain a set of gaps and to fill in those gaps pipelined read file operations, not more burst reads.  This handles lost bytes very efficiently.

IamPete1, 2023-01-15:
This should help once we end up doing one by one. https://github.com/ArduPilot/MissionPlanner/pull/3033

New param will be useful for testing!

tridge, 2023-01-20:
the recent change here made a big difference:
https://github.com/ArduPilot/MissionPlanner/commit/0e49cec25eee6c111b7cb6856eaf6565db3ee842
still not quite as fast as MAVProxy, but a lot better!

tridge, 2023-02-06:
latest changes make this fast enough

closed 2023-02-06 by tridge