ArduPilot/MissionPlanner #3053: Connect using Hostname

o-gent, 2023-02-04:
#### Issue details
Hostname connection not supported through UDPCI / TCP

I've checked previous issues for any mention, I've tried on MP and I've had a google - I could have missed a discussion or maybe the feature is already implemented in some way..

Currently need to specify IP - more convenient to specify hostname, especially where the IP of the device may be changing. For example the hostname would be "raspberrypi" by default for a pi.

#### Version
current

robertlong13, 2023-03-01:
Connecting via hostname works fine for me. For instance, I usually type `localhost` into that box instead of say `127.0.0.1`. Double check that the name you are trying actually resolves, e.g. open a command prompt and type `ping raspberrypi`. Usually it's `raspberrypi.local`, not just `raspberrypi`

o-gent, 2023-03-03:
Ah ".local" will be it then, I'm not used to needing that - thank you : )

o-gent, 2023-03-22:
[image]

Seems to fail with an ipv6 address? Works when manually using an ipv4, fails when manually pasting ipv6 too..

Image from when using hostname.local

o-gent, 2023-03-22:
from a look in UDPSerialConnect.cs, looks like it's setup only for ipv4 address - think the easiest way could be to filter to only ipv4 addresses from the Dns.GetHostEntry method in a similar way to here https://stackoverflow.com/questions/1059526/get-ipv4-addresses-from-dns-gethostentry ?

o-gent, 2023-03-22:
Confirmed that Dns.GetHostEntry(host).AddressList contains the ipv4 and ipv6 address for the machine i'm trying to connect to, so we need to filter to get the correct one rather than just picking the first address which is the current behaviour

meee1, 2023-03-22:
my quick google suggests your error is to do with your firewall
https://thegeekpage.com/address-incompatible-with-the-requested-protocol-10047-error-fix/

meee1, 2023-03-22:
also what OS?

o-gent, 2023-03-22:
I think it's as a result of using an ipv6 address 
[image]
I can try the other steps, but the connection works using the ipv4 address

o-gent, 2023-03-22:
> also what OS?

Windows 11

o-gent, 2023-03-22:
thanks for your response! @meee1

meee1, 2023-03-22:
ok, i think i might have an answer, just needs testing, well release a new beta soon

o-gent, 2023-03-22:
ok so I've tried the latest beta build 1.3.8479.1107 and now it just stays on trying to connect. Normal behaviour on ipv4 address

o-gent, 2023-03-23:
tried the 1.3.8482.6581 build with the ipv6 fix in the changelog @meee1, not really sure how to interpret the error, had a google and seems pretty generic.. 
- pinging the hostname in console responds with an ipv6 address
- manually entering the ipv6 address stays on "trying to connect"
- Using the ipv4 address works

[image]